Yeraze/meshmonitor #42: Info screen has bad node information

cipriani14, 2025-09-29:
I got this setup this morning and so far, it seems to be working fine. I think? 

I did notice on the info tab, however, that it’s showing wrong information about my Node. For example, it says it’s currently set to Medium Fast, when I have Long Fast enabled. It also details MQTT is active to a server I’ve never seen before, when in fact MQTT is turned off. 

[image]

Also — when connecting to the Node via the Meshtastic app to verify all of my settings, I received this error almost immediately: ⚠️ Client Notification: Multi-hop traceroute to broadcast address is not allowed

foehammer, 2025-09-29:
ditto. I'm not in US or on MediumFast, but I'm seeing the same.

Yeraze, 2025-09-30:
Doh.. Thanks for the heads up, on it.