Yeraze/meshmonitor #55: [FEAT] Support multiple nodes

Yeraze, 2025-09-30:
**Is your feature request related to a problem? Please describe.**
I have direct network access to multiple nodes and would like to monitor them all as part of the holistic mesh.  

**Describe the solution you'd like**
This can be solved right now by simply running multiple instances of the meshmonitor docker container, each one pointed at a single node.  This request is to have a single instance that allows me to monitor each node individually, as well as get a global holistic view across multiple nodes.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Just filing this here for people to Upvote or Comment on if they find it useful!

techniker, 2025-10-18:
Someone already looked at Malla(https://github.com/zenitraM/malla)?
Multiple node ingest via MQTT.

Littleaton, 2025-10-24:
Would love this option for managing/viewing remote nodes that have internet access.

jmrplens, 2025-11-06:
That would be a useful feature.
I have two MeshMonitors, one for LongFast and another for MediumFast.

If there could be a MeshMonitor server and then a MeshMonitor client for each node, several nodes could be centralised and even if one went down, the server would continue to function correctly.

wlockwood, 2025-12-07:
> I know a ton of users who would be willing to share their MeshMonitor data to create a mega dashboard and see all of the nodes for a whole city or something in one.

Presumably that wouldn't be vastly different from any of the many MQTT-based maps.

Yeraze, 2025-12-11:
Closing this issue as MeshManager now exists ... It's still pretty raw and not production-ready, but it's available for anyone who wants to try it out.